vedderb/bldc #666: [LispBM] Programmatically delaying the shutdown event

dcodeIO, 2023-10-28:
When using a power switch and listening for `event-shutdown` in LispBM, there is a hardcoded 100ms window to perform cleanup before the VESC is turned off. While useful to store encoder state or similar before turning off, I have a use case where I'd like to delay the shutdown until an action taking longer than 100ms is complete.

Ideally, I'd like to receive `event-shutdown` normally and be able to interrupt (in effect: delay) the shutdown process, finally calling, say, `(shutdown)` in LispBM once the action is complete. Is this functionality something worth considering? Or is this already possible somehow when using only LispBM (no custom firmware)?

vedderb, 2023-10-31:
I just added this one: https://github.com/vedderb/bldc/commit/71857f60986ea3ce3f822ce8222f8cf9df58e7f1
Do you think that works for you?

dcodeIO, 2023-10-31:
Thanks! Looks as if it should work for my use case. Looking forward to trying it out!

closed 2023-11-01 by vedderb

dcodeIO, 2023-11-06:
Works great!

dcodeIO, 2023-11-13:
Tried it out a litte more meanwhile, and found:

* When turning the power switch on again during `shutdown-hold true`, the VESC does not turn off as expected but the gate is not enabled again, so motors no longer function when setting duty etc. Iirc there is another timeout ([here](https://github.com/vedderb/bldc/blob/4d734d0c6e58c4ad3d1bbd9f34a34eaaec224800/hwconf/shutdown.c#L154-L164) perhaps?) for re-enabling the gate in this case that might have to take `shutdown-hold` into account now as well.
* When using `sleep` within the delayed shutdown event hook, execution occasionally fails with a rogue `read_error` (no source location except line 1, column 22 or so). Does not happen when using `yield` instead.