vedderb/bldc #667: [LispBM] Registering an extension that doesn't start with "ext-" silently passes

lukash, 2023-10-28:
Calling:
```C
VESC_IF->lbm_add_extension("my-extension", my_extension);
```

Passes without error, but it seems the `"ext-"` prefix is required, otherwise the extension is not registered. It should instead fail with an error.

tmgofer, 2023-10-30:
I confirm. I've encountered this issue in v. 6.02. Looks like "ext-" prefix is mandatory (will have to look in code to confirm, though).

vedderb, 2023-10-31:
In 6.05 it should not be mandatory when using upload, only when using stream. The reason for that is quite complicated and has to do with the non-incremental version of the reader. @svenssonjoel can explain better than me.

svenssonjoel, 2023-10-31:
As code (and extensions) can be loaded dynamically there is a risk that a symbol representing an extension is loaded before the extension itself is created and added. In these cases we need a way to syntactically determine if a symbol refers to an extension and this is why the "ext-" prefix was chosen. This scenario is most likely to happen when loading custom C libraries.

tmgofer, 2023-10-31:
> In 6.05 it should not be mandatory when using upload, only when using stream.
Won't this cause a confusion? Maybe it should be mandatory everywhere and raise an error when either:
- (load-native-lib ) is called (e.g. inside _VESC_IF->lbm_add_extension()_ )
- or during compilation of the native library (not that straightforward...)

vedderb, 2023-11-01:
Here is an attempt: https://github.com/vedderb/bldc/commit/0fb63f6019f6807234de61257f7cbee077d8ab4f
It will now return false and also print an error

closed 2023-11-01 by vedderb