vedderb/bldc #678: Better support for digital/ switch-based brakes

AkbarRamzan, 2023-11-17:
The problem with the adc app is that the ramping time affects both brake and throttle. I have an adc throttle but reed switch / digital brake and pressing the brakes causes the brake input to turn on too quickly. I want the regen brake to become stronger over time and ramp up more slowly but keep the throttle responsive. Can we have separate ramp up/ down times?

tmgofer, 2023-11-20:
So you have on/off switch on ADC2? Do i understand correctly?

it shouldn’t take more than 15 lines of LISP script…

AkbarRamzan, 2023-11-20:
Correct, yes but I prefer it done in the firmware as it safety-critical. What if you have a throttle runaway event and lisp crashes?

tmgofer, 2023-11-20:
Implement native C function, then. Just call it from LISP. I’ve done that and it works really well. 
Still, for such low computation functionality I’d suggest simply using LISP…