vedderb/bldc #682: Can 1 STM32F drive 2 mosfet Driver?

Knockoi, 2023-12-03:


ilbeylia, 2023-12-18:
This project allows you to control two DRV8301 MOSFET drivers using a single STM32F4 microcontroller through SPI. The STM32F4 series typically has two or more SPI interfaces. Here are the SPI pin connections:

You can use two SPI units, for example, SPI1 and SPI2.
Example pin connections for SPI1:

SCK: PA5
MISO: PA6
MOSI: PA7
NSS: PA4
Example pin connections for SPI2:
SCK: PB13
MISO: PB14
MOSI: PB15
NSS: PB12
You can configure these connections using the STM32CubeIDE and write the necessary code to complete your project. If you need assistance, feel free to reach out to me.

Knockoi, 2023-12-19:
Thank you very much for the information, I still need to connect the Hall
detector, voltage, current and other data lines so how do I do it? It seems
that the hwconfig file can only be used with one drv8301?