vedderb/bldc #706: Lisp app-adc-override wrong values

Nikolai521, 2024-02-09:
Hi, today i tried the new Beta 6.05 (2024-02-01)

I use my 75100 via lisp with the 1zuna script for my scooter, after update everything seems fine, but i lost the ADC values, they are VERY low, just 0.1 to 0.8v max, after this i tried the previous beta and the values go back to normal without issues.

Looking from changes i noticed some changes in the adc-override from 2 weeks, I bet this commit caused the issue https://github.com/vedderb/bldc/commit/c5b8bb7febfc6e910c16a51ea279d19a425d5d37

This is the values from the newer beta: 
[image]

And this one is from 31/12/23
[image]

Gabrielerusso, 2024-02-18:
I don't think there is any issue, the change was just to convert a value from 0 to 1 to simply cropping the value between 0 and 3.3, so you can't use the same code as before I think.

Nikolai521, 2024-02-18:
Probably, but now the resolution of these values are waaay to low

Gabrielerusso, 2024-02-19:
> Probably, but now the resolution of these values are waaay to low

i don't understand how after that change the resolution get lower. Post a code example that gives you a "lower resolution".

Gabrielerusso, 2024-02-19:
> now the resolution of these values are waaay to low
i don't understand how after that change the resolution get lower. Post a code example that gives you a "lower resolution".

The picture you posted are exactly the value you used before, but scaled from 1 to 3.3, 0.5306V * 1/3.3 = 0.160788 that rounded is 0.1608, that means you are giving the wrong input to the override function. Before it was a value between 0 and 1, now is between 0 and 3.3 .

Nikolai521, 2024-02-19:
Actually is 1zuna code, you can find them on his repo https://github.com/m365fw/vesc_m365_dash/tree/6_05_adc 

probably im going to record a video of the values, i cant explain well

Gabrielerusso, 2024-02-19:
> Actually is 1zuna code
Again it's not a vesc issue but the code you posted wasn't updated, don't record any video. If i can i will mache a branch and fix the @1zun4 repo.

Nikolai521, 2024-02-19:
Yea sure, also i tried to contact with him and i got no response, so some friends asked me to do it in the vesc repo

1zun4, 2024-02-19:
😩

Gabrielerusso, 2024-02-19:
https://github.com/Gabrielerusso/vesc_m365_dash/tree/6_05_beta_adc fix there

Gabrielerusso, 2024-02-19:
> Yea sure, also i tried to contact with him and i got no response, so some friends asked me to do it in the vesc repo

https://github.com/Gabrielerusso/vesc_m365_dash/commit/589a7bfe49eec6579431dae9225cf1b6ba0f3938 you can find here the fix

Nikolai521, 2024-02-19:
Now it makes sense! thanks c:

Gabrielerusso, 2024-02-19:
> Now it makes sense! thanks c:

Please report here, in my repo or in the Pull request if the fix actually works

Nikolai521, 2024-02-19:
> > Now it makes sense! thanks c:
> 
> Please report here, in my repo or in the Pull request if the fix actually works

[image]

Now i get more rasonable values, but the script crashes if release the controls, with the voltage stuck on the last value
also i noticed it crashes just at 0.9974v (very near to 1)
[image]
[image]

Gabrielerusso, 2024-02-19:
> > > Now it makes sense! thanks c:
> > 
> > 
> > Please report here, in my repo or in the Pull request if the fix actually works
> 
> ![image](https://private-user-images.githubusercontent.com/25958695/306011823-216c16a4-f087-4750-80b0-72dd96aa2d58.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgzNTg3NTgsIm5iZiI6MTcwODM1ODQ1OCwicGF0aCI6Ii8yNTk1ODY5NS8zMDYwMTE4MjMtMjE2YzE2YTQtZjA4Ny00NzUwLTgwYjAtNzJkZDk2YWEyZDU4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE5VDE2MDA1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzODc4OWM1MzdjZDg4ZWRhYTc2NjJjZDc3MzhkZmYwZGE0OTlmZjE5NWUyNmQ5Yzk3ZjllYTJhODkxODcxYWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.JTDyC3ZqQeiMbr-SL60XRRTEM_kUQ5qh6QHl-AAfS9k)
> 
> Now i get more rasonable values, but the script crashes if release the controls, with the voltage stuck on the last value also i noticed it crashes just at 0.9974v (very near to 1) ![image](https://private-user-images.githubusercontent.com/25958695/306012205-88b961fb-c306-4070-ab5c-d5dd92741079.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgzNTg3NTgsIm5iZiI6MTcwODM1ODQ1OCwicGF0aCI6Ii8yNTk1ODY5NS8zMDYwMTIyMDUtODhiOTYxZmItYzMwNi00MDcwLWFiNWMtZDVkZDkyNzQxMDc5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE5VDE2MDA1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk0NmYxNDRjMWUwNzY0ZjY1NGM5MDNlZmU2ZjhiMjQ4MjY1ZTcwNDExZjY2NTJmMDExNDJkM2NkOTI0NGQwZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.EGrWPrsUPoHxh34muiprlNAs_BDU8b5lFnMzrHwyEaQ) ![image](https://private-user-images.githubusercontent.com/25958695/306012325-015d89f9-1cc9-4f30-8091-d53b86d67773.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgzNTg3NTgsIm5iZiI6MTcwODM1ODQ1OCwicGF0aCI6Ii8yNTk1ODY5NS8zMDYwMTIzMjUtMDE1ZDg5ZjktMWNjOS00ZjMwLTgwOTEtZDUzYjg2ZDY3NzczLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE5VDE2MDA1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVhMjYyMzQ1ODcwMjAwZDI1Nzk2MjY5NjIyYzI0NWE0MjAxMzcxYmFlNTk5MzEzMjI4ZTNhNjllNmI3ZjkyOTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.YTE-RJuep9T3ZU0LD6xIme3le6oFKD12VCmenm1gntA)
> 
> edit: yep, it crashes just at 1, it works well under 0.9

are you using the scripts from my repo? what are you using (m365 or g30)?

Nikolai521, 2024-02-19:
> are you using the scripts from my repo? what are you using (m365 or g30)?

My error, now it works!
[image]

i copied bad some lines :skull:

Gabrielerusso, 2024-02-19:
> i copied bad some lines 💀

When someone asks "Are you using the script from my repo" means the whole script, as for future reference don't do copy/paste mix-up or it becames impossible to track an issue.

Nikolai521, 2024-02-19:
> Close the issue if it works.

Yup now works, actually it was my mouse, i moved it when i do the copy
Closing c:

closed 2024-02-19 by Nikolai521