vedderb/bldc #715: CAN bus spamming

sshenker85, 2024-03-07:
I have noticed the CANBUS being spammed with message ID 0x917 regardless of status message selection or update rate. Is this an intended function for a certain operating condition? Not sure why this is the case, please let me know if I am missing something.

Huangney, 2024-10-17:
Same my friend😭, have you solved this problem? If I don't set the option "CAN_Status_Message_Mode" as "CAN_Status_Disabled", My CAN bus is spammed with message EXTID 0x965 regardless of rate of other mode selections. I only knows that the 9 means status_1, and the 0x65 is the CAN_ID of my VESC Driver. BUT WHY CAN BUS WAS ALWAYS FULL, It annoyed me foy days.
[image]

sshenker85, 2024-10-17:
Yes! You need at least 1 reciever on the bus to acknowledge the CAN message.

Steven Shenker

Huangney, 2024-12-26:
Thanks :) , I solved that at the end of the Oct. I find that the VESC's message need to be replied immediately, or it'll keep sending messages. The way I solve the problem is to reply its message in the CAN_RECV_INTR function, it's fast enough.