vedderb/bldc #732: Feature request: updating c_libs interface header

tmgofer, 2024-06-12:
Hi,
Are you planning to extend https://github.com/vedderb/bldc/blob/master/lispBM/c_libs/vesc_c_if.h and publish more functionality that will be accessible to c_libs? There are many functions that can be invoked in LISP, but are not published for native libraries (eg. foc-beep calls mcpwm_foc_beep, which is not exposed in vesc_c_if.h. Also, just a few settings can be set/get by VESC_IF->set_cfg_float)
Please consider extending the interface.

tmgofer, 2024-06-12:
I've forgot to mention - I'm happy to implement the update on a branch if there's a chance it will be included in next release.

vedderb, 2024-06-17:
I did not plan to spend a lot more time on vesc_c_if before the release and I did not intend to make it support everything that is supported by the lisp-extensions. As the c interface has harder restrictions on binary type compatibility it is harder to test and maintain, and making changes to it is more likely to break old code using it. The firmware is also getting quite large for the flash size and I don't want to use the last space for duplicating functionality.

If there is anything in particular you need for the C interface we can add that. If you just want to have access to more functions in general you can interact with lisp from your C code and use the lisp-funcctions in a few different ways.

tmgofer, 2024-06-18:
Thanks for answer, Benjamin. I understand the effort related to maintenance and hazard with keeping binary compatibility. Makes sense.
You're right, I wanted to have access to more functions in general -  that's because of performance issue I've experienced in LISP in a more complex applications. 
I wasn't aware that I can call a LISP function from within c_lib function. Any hint on how to do that?