vedderb/bldc #740: Get ESC faults via CAN

drewskoots, 2024-07-02:
Is there a way to get faults from an ESC via the CAN network? It seems that VESC tool is able to do so, but in the VESC CAN documentation, it isn't documented anywhere.

DovPearX, 2024-07-08:
You can send a command via CAN to download data and the entire structure along with error codes will be there. Exactly the same way as in the vesc uart libraries

DovPearX, 2024-07-08:
You can send a command via CAN to download data and the entire structure along with error codes will be there. Exactly the same way as in the vesc uart libraries.

manmountain, 2024-07-15:
What can command to use for requesting fault codes? I cant find anything like that from datypes.h CAN_PACKET_ID enum

DovPearX, 2024-07-15:
When using CAN, fill the receive buffer with the COMM_GET_VALUES command. Of course, you also need to make sure that the device you are sending from can receive the data in the commands.c file.

void comm_can_get_values(uint8_t controller_id) {

    int32_t send_index = 0;

        uint8_t buffer[4];

    buffer[send_index++] = COMM_GET_VALUES;

    comm_can_send_buffer(controller_id, buffer, send_index, 3);

}

spamsch, 2025-07-11:
To test this you might want to have a look at https://github.com/waas-rent/vesc_can_sdk. There is a shell that can be used to quickly check.