vedderb/bldc #749: MakerX, Flipsky: warning about damage when using current official firmware

c0d3z3r0, 2024-07-25:
When looking at VESC-compatible controllers, I found two warnings about official firmware:

[image]
https://www.makerx-tech.com/pages/about-phase-filter

[image]
https://flipsky.net/collections/v4-series/products/torque-esc-vesc-%C2%AE-bldc-electronic-speed-controller

I guess this is related to phase filters as well, since it's based on VESC 4.12, which lacks them.

Now, I wonder
1) is this still an issue?
2) what is actually causing the damage? As both don't provide schematics, I have no clue....
3) Is there anything that can be done, to *prevent* damage?

TechAUmNu, 2024-07-25:
All of these escs have the same 'sensing' hardware (except phase filters) as the 75300 config, and they all for some crazy reason use that config. Problem is the models all use completely different fets and layouts that might not actually handle the same power as a 75300. They apparently don't have anyone at these companies that can do firmware, so they just keep using what works and recommend not updating. They even have the Flipsky 75450 which can't even sense above 495A. It's diabolical. 

There are community firmware configs added for most of the china escs so those should be used instead if you want to update to the latest firmware. 

So
1. Yes
2. No phase filters are fitted, so when it tries to use them there is no filtering on the signals and it can't track the motor at all, tbh this wouldn't be a problem if they didn't use nonsensical configs with default limits way to high for their boards....
3. Use the provided configs for the esc you have, not the default 75300 one that comes with them, or just turn off phase filters.
Really this is THEIR problem and THEY need to fix it by submitting a PR with sensible configs for all their hardware versions. Unfortunately I don't think that will happen anytime soon.

c0d3z3r0, 2024-07-26:
Thank you!

>    Really this is THEIR problem and THEY need to fix it by submitting a PR with sensible configs for all their hardware versions. Unfortunately I don't think that will happen anytime soon.

Yeah, fully agreed. Sounds like the usual chinese clone crap... I'm currently reverse engineering Qiroll PRO v2 (check out https://github.com/c0d3z3r0/qiroll_rev). Guess what they used? VESC 4.10, fw 4.00, GPL violated, settings completely wrong.......................

> All of these escs have the same 'sensing' hardware (except phase filters) as the 75300 confi

What du you mean with `'sensing' hardware`, exactly? I had seen some forum post somewhere about "no real sensing" but without any details :/

Regarding phase filtering, it looks like at least Flipsky has added phase filtering now (FS Mini V6 MK5 - clone of V6 MKVI I guess, and FS 75200 Pro V2.0)

By the way, I'm completely confused of all these versions... do you happen to know what exactly VESC 6.6 is, and what the differences between MKV and MKVI really are, as there don't seem to be any schematics for 6.6 and MKVI...?

c0d3z3r0, 2024-07-27:
@TechAUmNu sorry for bothering. If you have any more details, it would be awesome if you'd share them :-) Would help a lot!

TechAUmNu, 2024-07-27:
By sensing hardware I mean the voltage dividers on the phase and bus voltage, current shunt resistance and current amplifier gain. E.g. some of the lower current flipsky etc stack 3 shunts on top of each other so the resistance is the same as the 75300 config. Which means there is way more current noise than their should be. You can also then set current values that for sure will damage the mosfets etc. 

The whole naming thing for those china vescs is a complete mess honestly. There isn't a defined 6.6 means "this", it's just what they called it. 

So the main differentiators that actually matter are these
2 or 3 Low side shunts (no current filtering) + with/without phase filters
2 or 3 Phase shunts (with/without current filtering) + with/without phase filters
Current rating of mosfets
Voltage rating of mosfets, caps, etc.
Hardware protection (comparators for overcurrent, etc)
Construction of esc (heatsinking, metal core pcb, input and output wiring, connector types)

Any other differences are usually more user experience things like push to start, loggers, anti-spark, etc.

c0d3z3r0, 2024-07-27:
Thank you so much, this helps a ton!;

> By sensing hardware I mean the voltage dividers on the phase and bus voltage, current shunt resistance and current amplifier gain. 

Ah, I interpreted that 'sensing' as ironic :-) there was some comment in some forum that they don't have 'real' sensing.

> E.g. some of the lower current flipsky etc stack 3 shunts on top of each other so the resistance is the same as the 75300 config. Which means there is way more current noise than their should be. You can also then set current values that for sure will damage the mosfets etc.

OMG what a mess

> 
> The whole naming thing for those china vescs is a complete mess honestly. There isn't a defined 6.6 means "this", it's just what they called it.

Ah ok

> 
> So the main differentiators that actually matter are these 2 or 3 Low side shunts (no current filtering) + with/without phase filters 2 or 3 Phase shunts (with/without current filtering) + with/without phase filters.

Ah yeah, so that's basically VESC 4 vs. 6 and the ones depending on 6 like MKV, MKVI, and the 75**** ones, right?

> Current rating of mosfets Voltage rating of mosfets, caps, etc.

> Hardware protection (comparators for overcurrent, etc)

Hm, didn't see any comparators for protrction, yet, what model has them?

> Construction of esc (heatsinking, metal core pcb, input and output wiring, connector types)
> Any other differences are usually more user experience things like push to start, loggers, anti-spark, etc.

Any idea what the "MAX" for Trampa MKVI means?

c0d3z3r0, 2024-07-27:
Hahahaha there's a picture of these stacked shunts, omg wtf :'D https://forum.esk8.news/t/how-to-update-firmware-on-the-flipsky-75100-75200-foc-esc/61819

closed 2024-08-15 by c0d3z3r0