vedderb/bldc #751: Confgenerator license

mtboswell-aeroseal, 2024-07-29:
Would it be possible to obtain the header files which are required to use the confgenerator.c file generated the VESC tool under a license other than the GPL? We would like to implement software to configure a VESC device without requiring our application to be GPL’d as well.

vedderb, 2024-08-02:
Which header file are you referring to?

mtboswell-aeroseal, 2024-08-02:
confgenerator.h includes datatypes.h, which then includes a bunch of other stuff. At the top of datatypes.h and the files it includes it states the license as GPL3. Thanks for taking a look at this.

mtboswell-aeroseal, 2024-08-09:
Was there an intent for other software to be able to configure a VESC that way? Or was that feature intended for something else?