vedderb/bldc #763: Regarding setup for uploading firmware

Shubham260202, 2024-09-12:
I have get the pcb from the website blog of 2015. which version was 4.12 in the blog it redirected to this git repository and i tried to follow the step given in readme file also in the blog below. i have tried the steps both in ubuntu and windows. In ubuntu i am not getting access to launchpad and in windows PS C:\Windows\system32\bldc> make arm_sdk_install
make: *** No rule to make target 'arm_sdk_install'.  Stop. This is the error i am getting how to solve it and how to upload the firmware and bootloader to the pcb. Please give me some guidance to it

[image]

jaykup26, 2024-09-12:
If you don't have a need to compile, you could always load the binary firmware file directly with an stlink v2.

Use the vesc_default.bin from here

https://github.com/vedderb/vesc_fw_archive/tree/main/6.05/410_o_411_o_412

Once the firmware is loaded and your are able to connect, you can load the bootloader from the vesc tool.

Detailed stlink instructions

* Hook an ST-LINK v2 up to the SWD port. Only use the GND, DIO and CLK pins. Have the battery plugged in. If you cannot plug in to the battery, then use the 3.3v pin as well, but this is not the recommended method as it can sometimes kill the 5v to 3.3v buck converters. 
* Plug the ST-LINK v2 into the PC
* Download the [STM ST-LINK Utility](https://www.st.com/en/development-tools/stsw-link004.html)
* Open and click Target → Connect.
* Click Target → Program & Verify
* Click Browse and load the .bin firmware file.
* Click Start.
* Wait 30 seconds then unplug the STLINK and the battery from the ESC. Plug in the ESC to the battery and connect through the normal USB port or bluetooth to configure with the VESC tool or VESC app.

Shubham260202, 2024-09-12:
I tried this method but it disable my port from the hardware have you verified the steps

jaykup26, 2024-09-15:
Yes, I have done this many times when replacing STM32 chips. New chips obviously have no firmware/bootloader on them.  410_o_411_o_412 is the correct firmware for the design above.  STM32CubeProgrammer is the new software to do this, but the ST-LINK Utility still works fine.

You may find more help on this issue in the VESC Discord, it's pretty active.  Seems like there may be hardware issues if the ST-LINK won't connect.  https://discord.gg/JgvV5NwYts