vedderb/bldc #777: Can't read motor configuration on FW 6.06

TrovaD, 2024-11-01:
I have a FSESC V 6.7 and currently, VESCT Tool can't read or write motor configurations. I forked a version in July, this worked well until i updeatet it two weeks ago. I had a look into the whole morning, but couldn't figure out how to solve it. With 6.05 it works well

amper128, 2024-11-06:
Same problem, I use FW 5.x for FSESC.
On FW 6 motor and app configuration cannot read and cannot modified.

Alec1of1, 2025-01-25:
Absolutly same problem, not reading motor configuration by Vesc Tool 6.05 and 6.06 after update from repasitory FW 6.06

Alec1of1, 2025-01-27:
Can be marked as solved 
Main problem are not follow the instructions from README.md.,
downloaded zip file is testing FW "bldc-master"  branch going to throw error on motor configuration.
To ovoid this for common user must must be perform instructions from page
https://github.com/vedderb/bldc/blob/master/README.md

as for Linux

git clone http://github.com/vedderb/bldc.git

folder name will be "bldc"     as stable                          and not "bldc-master     as testing

successfully build  with       /tools/gcc-arm-none-eabi-7-2018-q2-update