vedderb/bldc #788: Feature request: turn off the motor on thermistor failure

l29ah, 2024-12-06:
I have a 10k NTC thermistor installed in the motor and attached to VESC 4.12. When it gets disconnected, VESC tool reports wildly negative temperature, yet doesn't prevent one from using the motor in any way, potentially burning it if the sensor is accidentally disconnected during a ride.

It would be great to severely limit the motor current or completely disable the motor until the sensor connection is fixed if the sensor reports abnormally low temperature.

jaykup26, 2024-12-06:
If implemented, this should be a separate option in motor -> general -> temperature that is turned off by default.

Many VESC users are using motors without temperature sensors or need to keep motors operational even if a temp sensor is disconnected (electric skateboards, one wheels, EUCs, etc) and this could cause a lot of usability problems.

l29ah, 2024-12-06:
Band-aid solution that just triggers a fault when it's too cold:
```
diff --git a/motor/mc_interface.c b/motor/mc_interface.c
index a2e3519d..f4a2be56 100644
--- a/motor/mc_interface.c
+++ b/motor/mc_interface.c
@@ -2330,6 +2330,9 @@ static void update_override_limits(volatile motor_if_state_t *motor, volatile mc
        float lo_max_mot = l_current_max_tmp;
        if (motor->m_temp_motor < (conf->l_temp_motor_start + 0.1)) {
                // Keep values
+               if (motor->m_temp_motor < -30) {
+                       mc_interface_fault_stop(FAULT_CODE_OVER_TEMP_MOTOR, !is_motor_1, false);
+               }
        } else if (motor->m_temp_motor > (conf->l_temp_motor_end - 0.1)) {
                lo_min_mot = 0.0;
                lo_max_mot = 0.0;
```