vedderb/bldc #793: Power limits issue/improvement: Field weakening injected current are not limited by max power limits

ljmf00, 2024-12-27:
The overall power draw limits doesn't take into consideration the injected current by Field Weakening parameters (probably some other injected current by other parameters), so if the overall power applied to the motor phases excluding Field Weakening is 950W and the limit is 950W, adding 200W by injecting current with Field Weakening, would theoretically draw 1150W.
 
The issue here is that the power limits doesn't cut Field Weakening injected current and the actual power goes beyond the specified limit.

If the current setting to limit power is designed to exclude Field Weakening, would be cool to add another extra limit for the overall power limit (including any injected current apart from the normal current applied to each phase).

This is useful to control how much power the motor gets. My motor is constructed in a way that is very much optimized for torque rather than speed, so I have to rely on Field Weakening to increase speed. I verified situations where Field Weakening is very effective on downhills but not on uphills, so, limiting the injected power of Field Weakening on uphills when power is already at motors maximum capabilities is desired. I also verified that the effectiveness directly relates to source of heat in the motor, so if Field Weakening isn't being effective, it increases the motor heat substantially.

For example: I'm in a uphill with the limit of 950W and Field Weakening kicks in, then, I'm getting power beyond 950W and causing excessive power, where Field Weakening is not effectively doing much, so, instead, that power is being transferred in form of heat rather than being efficient at increasing speed. Limiting the overall power by preferring phase current rather than Field Weakening injected current, would definitely help on reducing the heat.

My configuration doesn't include a sensor/thermistor in the motor for measuring/controlling/throttling temperature, so, the only way to control/throttle it is indirectly via the overall power.

ljmf00, 2024-12-27:
Moreover, I verified that the battery current limits are not respected when Field Weakening is used. This correlates to the same issues but instead with current limits rather than power limits. Improving limits here also improves VESC safety as BMS wouldn't cut off if someone sets field weakening to dangerous values, so current would be within the limits too.

vedderb, 2024-12-27:
The problem is that there is no straight forward way of limiting the input current or power with field weakening or MTPA as the task becomes recursive. However, there is an active regulator that can be used to attempt that. You can enable it by setting **Input Current Limit Map Start** to something like 85%

[image]

ljmf00, 2024-12-27:
> The problem is that there is no straight forward way of limiting the input current or power with field weakening or MTPA as the task becomes recursive. However, there is an active regulator that can be used to attempt that. You can enable it by setting **Input Current Limit Map Start** to something like 85%. Edit: I think this should also respect the power limits, but I'm not sure as I haven't tested. Normally I never set a power limit.

I see, I'll try that and give some feedback here. From what I understand, this is an approximation to what I'm describing as a solution? It means the current starts to be limited towards the maximum input current, but can still go beyond it? Effectively remapping the FW and MTPA current as it goes beyond and then regulating itself as it maintains a high value? Does this option prioritize weakening the FW and MTPA current values over phase current? Would be good to be able to tweak that, otherwise.

Will close this ticket and reopen if I still see room for improvement here. Will try to plot some values to understand the impact of this.

closed 2024-12-27 by ljmf00

vedderb, 2024-12-28:
You are describing a problem, not a solution. If I could simply make the current limit account for field weakening and MTPA I would have done so already, but there is no straight forward way of doing that. The setting I described is the best thing I have come up with so far for solving this problem. It is not perfect.