vedderb/bldc #833: In desperate need of help

JohEngstrom, 2025-06-22:
I'm trying to use COMM_SET_MCCONF_TEMP or COMM_SET_MCCONF_TEMP_SETUP over UART to dynamically apply power and speed limits to a Spintend UBOX V2 (dual VESC, FW 6.05). The goal is to switch between temporary profiles for different ride modes using an ESP32-S3 connected via UART.

Hardware:
Controller: Spintend UBOX V2 Dual VESC

Firmware: 6.05

Throttle: Connected via ADC Adapter V3

ESP32-S3: Connected over UART to master VESC

VESC Tool App: Set to “ADC and UART”

ADC Control Type: “Current No Reverse Brake ADC2”

What works:
I can send COMM_GET_MCCONF and parse the data correctly.

I can send COMM_SET_MCCONF_TEMP_SETUP (with calculated ERPM from wheel diameter), and I get back the correct 0-byte reply.

I’ve verified that the UART commands are built correctly (payload, CRC, etc.).

Using stored profiles (via VESC Tool), the limits do apply even when using the ADC adapter.

What doesn't work:
When sending COMM_SET_MCCONF_TEMP_SETUP or COMM_SET_MCCONF_TEMP, no speed or power limits are applied — the throttle is connected via the ADC adapter.

I’ve tried:

With forward_can = true and false

With ack = true and false

Sending the command multiple times

COMM_SET_CURRENT and COMM_SET_DUTY also do nothing, even in UART-only mode.

Questions:
Is COMM_SET_MCCONF_TEMP supposed to apply limits for ADC-based throttle sources?

If not, is there any workaround other than using stored profiles or rewiring the throttle?

Is there any known restriction that prevents ADC mode from respecting these temp limits?

Thanks for the amazing work and firmware. I'd appreciate any help debugging or clarifying what’s expected behavior here!

JohEngstrom, 2025-06-22:
Here is the code I'm using. Can anyone see what I'm doing wrong?

```
bool vesc_send_temp_profile(const vesc_temp_profile_t *profile, bool forward_can, bool ack, bool divide_by_controllers) {
  if (!profile) {
      ESP_LOGE(TAG, "Profile pointer is null.");
      return false;
  }

  ESP_LOGI(TAG, "Sending TEMP profile:");
  ESP_LOGI(TAG, "  forward_can: %s", forward_can ? "true" : "false");
  ESP_LOGI(TAG, "  ack: %s", ack ? "true" : "false");
  ESP_LOGI(TAG, "  divide_by_controllers: %s", divide_by_controllers ? "true" : "false");

  ESP_LOGI(TAG, "  l_current_min_scale: %.3f", profile->l_current_min_scale);
  ESP_LOGI(TAG, "  l_current_max_scale: %.3f", profile->l_current_max_scale);
  ESP_LOGI(TAG, "  l_min_erpm: %.3f", profile->l_min_erpm);
  ESP_LOGI(TAG, "  l_max_erpm: %.3f", profile->l_max_erpm);
  ESP_LOGI(TAG, "  l_min_duty: %.3f", profile->l_min_duty);
  ESP_LOGI(TAG, "  l_max_duty: %.3f", profile->l_max_duty);
  ESP_LOGI(TAG, "  l_watt_min: %.3f", profile->l_watt_min);
  ESP_LOGI(TAG, "  l_watt_max: %.3f", profile->l_watt_max);
  ESP_LOGI(TAG, "  l_in_current_min: %.3f", profile->l_in_current_min);
  ESP_LOGI(TAG, "  l_in_current_max: %.3f", profile->l_in_current_max);

  uint8_t payload[64];
  int32_t idx = 0;

  payload[idx++] = 0; // store = false
  payload[idx++] = forward_can ? 1 : 0;
  payload[idx++] = ack ? 1 : 0;
  payload[idx++] = divide_by_controllers ? 1 : 0;

  buffer_append_float32_auto(payload, profile->l_current_min_scale, &idx);
  buffer_append_float32_auto(payload, profile->l_current_max_scale, &idx);
  buffer_append_float32_auto(payload, profile->l_min_erpm, &idx);
  buffer_append_float32_auto(payload, profile->l_max_erpm, &idx);
  buffer_append_float32_auto(payload, profile->l_min_duty, &idx);
  buffer_append_float32_auto(payload, profile->l_max_duty, &idx);
  buffer_append_float32_auto(payload, profile->l_watt_min, &idx);
  buffer_append_float32_auto(payload, profile->l_watt_max, &idx);
  buffer_append_float32_auto(payload, profile->l_in_current_min, &idx);
  buffer_append_float32_auto(payload, profile->l_in_current_max, &idx);

  uint8_t tx_buf[96];
  size_t tx_idx = 0;

  tx_buf[tx_idx++] = 0x02;                     // Short packet start
  tx_buf[tx_idx++] = idx + 1;                  // Payload length + command
  tx_buf[tx_idx++] = COMM_SET_MCCONF_TEMP;     // Command byte

  memcpy(&tx_buf[tx_idx], payload, idx);
  tx_idx += idx;

  uint16_t crc = crc16(&tx_buf[2], idx + 1);
  tx_buf[tx_idx++] = crc >> 8;
  tx_buf[tx_idx++] = crc & 0xFF;

  uart_platform_flush();
  ESP_LOG_BUFFER_HEX_LEVEL(TAG, tx_buf, tx_idx, ESP_LOG_INFO);
  bool ok = uart_platform_write(tx_buf, tx_idx);

  if (!ok) {
      ESP_LOGE(TAG, "UART write failed.");
      return false;
  }

  uint8_t reply_buf[64];
  int reply_len = uart_platform_read(reply_buf, sizeof(reply_buf), 1000);
  ESP_LOGI(TAG, "Reply (%d bytes):", reply_len);
  for (int i = 0; i < reply_len && i < 16; ++i) {
      printf("%02X ", reply_buf[i]);
  }
  if (reply_len > 16) printf("...\n"); else printf("\n");

  return reply_len > 0;
}
```

vesc_temp_profile_t looks like this:

```
typedef struct {
  float l_current_min_scale;
  float l_current_max_scale;
  float l_min_erpm;
  float l_max_erpm;
  float l_min_duty;
  float l_max_duty;
  float l_watt_min;
  float l_watt_max;
  float l_in_current_min;
  float l_in_current_max;
} vesc_temp_profile_t;
```

spamsch, 2025-07-11:
This is not a bug or issue with VESC. Please ask this question on discord. We will be happy to help you there.